XenGi/teensy_library #1: Teensy 2.0 symbol doesn't fit foortprint

mediaartdesign, 2016-10-18:
The fact that you left out the second GND and 5V Pin in the symbol of Teensy 2.0 leads to a fatal shifting of pins when using your footprint. From pin 15 on the connection will not be correct.

If you don't want do put in extra pins on your symbol for multiple power pins (which I strongly recommend, though), you have to have multiple pads of the same number. Otherwise Kicad will give you an error when importing the netlist, because the number of pads does not match the number of pins.

corrected library symbol:

```
#
# Teensy2.0
#
DEF Teensy2.0 U 0 40 Y Y 1 F N
F0 "U" 0 1250 60 H V C CNN
F1 "Teensy2.0" 0 1075 60 H V C CNN
F2 "" 100 -1050 60 H V C CNN
F3 "" 100 -1050 60 H V C CNN
DRAW
S -500 1150 500 -1150 0 1 0 N
X GND 1 -700 1050 200 R 50 50 1 1 I
X 0 2 -700 900 200 R 50 50 1 1 I
X 1 3 -700 750 200 R 50 50 1 1 I
X 2 4 -700 600 200 R 50 50 1 1 I
X 3 5 -700 450 200 R 50 50 1 1 I
X 4_PWM 6 -700 300 200 R 50 50 1 1 I
X 5_INT0_PWM 7 -700 150 200 R 50 50 1 1 I
X 6_INT1 8 -700 0 200 R 50 50 1 1 I
X 7_INT2_RX 9 -700 -150 200 R 50 50 1 1 I
X 8_INT3_TX 10 -700 -300 200 R 50 50 1 1 I
X 13_A8 20 700 -300 200 L 50 50 1 1 I
X 24 30 -700 -925 200 R 50 50 1 1 I
X 9_PWM 11 -700 -450 200 R 50 50 1 1 I
X 14_A7_PWM 21 700 -150 200 L 50 50 1 1 I
X AREF 31 -700 -825 200 R 50 50 1 1 I
X 10_PWM 12 -700 -600 200 R 50 50 1 1 I
X 15_A6_PWM 22 700 0 200 L 50 50 1 1 I
X 23 13 -300 -1350 200 U 50 50 1 1 I
X 16_A5 23 700 150 200 L 50 50 1 1 I
X +5V 14 -150 -1350 200 U 50 50 1 1 I
X 17_A4 24 700 300 200 L 50 50 1 1 I
X GND 15 0 -1350 200 U 50 50 1 1 I
X 18_A3 25 700 450 200 L 50 50 1 1 I
X RST 16 150 -1350 200 U 50 50 1 1 I
X 19_A2 26 700 600 200 L 50 50 1 1 I
X 22_A11 17 300 -1350 200 U 50 50 1 1 I
X 20_A1 27 700 750 200 L 50 50 1 1 I
X 11_A10_LED 18 700 -600 200 L 50 50 1 1 I
X 21_A0 28 700 900 200 L 50 50 1 1 I
X 12_A9_PWM 19 700 -450 200 L 50 50 1 1 I
X +5V 29 700 1050 200 L 50 50 1 1 I
ENDDRAW
ENDDEF
```

XenGi, 2016-10-18:
Can you create a pull request so I can merge it easily? Thx for fixing!

XenGi, 2016-10-18:
Should be fixed with 47c9d194f32d146d093d0f810cc7d099cce29938.

closed 2016-10-18 by XenGi