XenGi/teensy_library #5: Missing pins importing netlist

NevynUK, 2017-04-03:
I have added the Teensy 3.6 symbol to a schematic, connected it up and then generated the netlist.

Now I read the netlist into the PCB layout too and I receive errors stating the footprints 57-62 could not be imported.

XenGi, 2017-04-04:
As you can see in this pinout the Teensy 3.5 and Teensy 3.6 have 5 Pins (58, 59, 60, 61, 62) marked `GND`, `GND`, `DP`, `DM` and `5V` with very odd positions. I'm not sure how to connect those. I'll have to check first if its actually possible to put a pin header on a PCB like that. If that fits I'll add the Pins. For now you can just ignore them if you don't need them. I'll update the footprint asap.

![teensy 3.6 pinout][pinout]

[pinout]: https://www.pjrc.com/teensy/card9b_rev1.png

XenGi, 2017-04-04:
Try debbe69f7c8bdf1deb3031f60854764a8d3d274a. This should fix it.

closed 2017-04-04 by XenGi