XenGi/teensy_library #16: Mainlining?

dogtopus, 2019-04-03:
Are you interested in adding this to the official [symbol](https://github.com/KiCad/kicad-symbols) and [footprint](https://github.com/KiCad/kicad-footprints) collection? It would need some changes to the look of the symbols and perhaps footprints (as documented here: http://kicad-pcb.org/libraries/klc/) but would make it much easier for people to access them and also integrates well with other symbols/footprints provided in KiCad.

XenGi, 2019-04-07:
Sure. I'll try to update it in the next week to meet the requirements. :D

dogtopus, 2019-06-17:
Hi are you still working in this? I am planning to draw a KLC-compliant teensy LC symbol for my own use (and will likely send a PR to KiCad) and I would like to ensure that no duplicate effort was made.

XenGi, 2019-06-17:
I had no time to work on it yet. Maybe you can just update my library and make it compliant? I could give you access rights to the repo. So you don't waste you time starting from the beginning.

dogtopus, 2019-06-18:
I think this would work.

XenGi, 2019-06-18:
I added you to the repo. Just send me a message if you need something.

XenGi, 2022-09-23:
So, KiCAD has a plugin manager now. How to you make something available in there?