XenGi/teensy_library #18: Teensy 4.0 is now available.

JayShoe, 2019-08-11:
Teensy 4.0 is now available. There are many changes from previous layouts. 

https://www.pjrc.com/teensy-4-0/

blackketter, 2019-08-12:
Anybody working on this?  I'm happy start on the T4 files, but don't want to step on anybody's toes.

blackketter, 2019-08-12:
See pull request  https://github.com/XenGi/teensy_library/pull/19

XenGi, 2019-08-13:
Symbol is added. Footprint comes when I got some sleep.

blackketter, 2019-08-13:
Thanks!  When you are rested, I'd love for another set of eyes on this and any feedback you may have.

JayShoe, 2019-08-13:
Hello friends, 

The footprint seems to have some options as being discussed on the forum. The pads underneath are accessed via a few different ways. 

 - Pogo pins
 - SMD header
 - Ribbon Cable
 - Flex PCB

I'm assuming the initial design should use pogo pins. But I'm interested in learning a little about the ribbon and flex PCB options, and to learn how it might affect the footprint.

blackketter, 2019-08-13:
Yes, I’m working on a footprint right now with castellated edges that should connect all the pins.

Do you think we should have two symbols, one for just the through-hole connections that can be done with headers and another that includes all the pins?

blackketter, 2019-08-13:
Here's where the footprint stands now: 

[image]

JayShoe, 2019-08-13:
Hello blackketter

> Do you think we should have two symbols, one for just the through-hole connections that can be done with headers and another that includes all the pins?

Sure why not? We might even want a few different options including the flex/ribbon cable option... And the Pogo options. 

> Here's where the footprint stands now:

Ahh! I see what you are saying about having the castellated edges... This is a good idea for after prototyping but for prototyping I think we will want removable parts.  We could have a few footprints. 

 - castellated (as you've drawn)
 - through hole only (as you've suggested, just the outer edge pins)
 - pogo'd / smd through holes ( I suspect the same layout will work with both of these options? When I say SMD version I mean this one. [http://media.digikey.com/Photos/Samtec%20Photos/TSM-107-02-S-DV-A-007_sml.JPG](http://media.digikey.com/Photos/Samtec%20Photos/TSM-107-02-S-DV-A-007_sml.JPG))
 - ribbon (there are two versions brewing. ribbon and flex PCB. The flex PCB looks promising but I haven't found it's receiving part yet.)

Sorry, not trying to say you need to do all of this, but just trying to help think through what might make sense.

blackketter, 2019-08-13:
Great.  

So, I think the through-hole version I already committed should be updated to include the holes for pins D24-D33 for the SMT header you referenced above.  

I think maybe there should be a separate version that also adds the pogo pins for the non-through-hole connections. (USB D+- and uSD connector).  Most folks won't need those holes and they might make routing harder for those cases.

I'll commit my castellated version and continue to refine it.  

I'm working on one or two breakout boards that use that footprint.  One that will bring out all the pins to 0.1" headers to make a removable CPU module.  

[image]

And another with castellated edges so that it can be mounted directly to a PCB.  This has the benefit of being able to mount the Teensy board level with main PCB and having the castellated breakout board act as a connector.

[image]

XenGi, 2019-08-13:
This looks pretty cool. For me my initial footprint is completly enough but feel free to put some merge requests in.

closed 2020-01-10 by XenGi