XenGi/teensy_library #29: 4.1 production vs Beta

DylanBrianBenton, 2020-06-17:
Ive noticed the 4.1 pins have changed from Beta from production. I'll verify it but if you want i'll fork it and you can update it if you'd like?

blackketter, 2020-06-17:
Can you elaborate on how they have changed?

DylanBrianBenton, 2020-06-18:
Im more than happy to be wrong. Looking at  the symbol [here](https://forum.pjrc.com/attachment.php?attachmentid=20624&d=1592409223) and the official pinout [here](https://forum.pjrc.com/teensy41_card.png) it looks like the pin descriptions for the SPI pins between 33-41 looks to be wrong. Happy to be wrong though :)

blackketter, 2020-06-18:
Good eye!  I didn’t see that.  I’ve made the changes and will commit. Please check my work.  Thanks!

closed 2020-06-18 by blackketter