XenGi/teensy_library #35: Teensy.lib flawed, Teensy.dcm not recognized

f1gwr, 2021-06-16:
KiCad version 5.1.6-0 Mac OS X 10.13.6
Hi,
Thanks for your fantastic job.
However:
The said library cannot be opened, see:
[image]
Yet while opening the file with a text editor, you can see that the github page text is also contained in the file. I tried to remove the parasitics and save it as .lib file, but KiCad doesn't accept it neither.

Even worse for teensy.dcm that KiCad can't even recognize, despite the fact 5.1 is still supposed to be able to access this deprecated format, officially abandonned in version 6: https://forum.kicad.info/t/is-dcm-file-will-be-deprecated/13763/5

blackketter, 2021-06-16:
How did you download the lib file?  Make sure you use the "Download Zip" option from the main page https://github.com/XenGi/teensy_library

XenGi, 2021-06-17:
I closed your other ticket in teensy.pretty because it seems like it is the same issue as this one. Let's discuss this in only one place.

closed 2022-05-24 by XenGi