XenGi/teensy_library #36: strange errors in kicad

folkertvanheusden, 2021-09-24:
Error: U5 pad 45 not found in teensy:Teensy40.
Error: U5 pad 46 not found in teensy:Teensy40.
Error: U5 pad 48 not found in teensy:Teensy40.
Error: U5 pad 50 not found in teensy:Teensy40.
Error: U5 pad 51 not found in teensy:Teensy40.
Error: U5 pad 52 not found in teensy:Teensy40.
Error: U5 pad 53 not found in teensy:Teensy40.
Error: U5 pad 54 not found in teensy:Teensy40.
Error: U5 pad 49 not found in teensy:Teensy40.
Error: U5 pad 47 not found in teensy:Teensy40.

blackketter, 2021-09-24:
I think you are using the through-hole footprint for the T4.0.  That footprint doesn't have pads for those pins.  The SMT footprint does.

folkertvanheusden, 2021-09-24:
I am indeed. So I can ignore this?

blackketter, 2021-09-24:
Yes, unless you need to use those signals!

We could make a second symbol that doesn't have those pins available, but I'm not sure that's necessary...

folkertvanheusden, 2021-09-24:
Or maybe a note in the readme that it is harmless if...

closed 2021-09-24 by folkertvanheusden