XenGi/teensy_library #38: SMD pads for 4.1 are not listed

mrwm, 2021-11-25:
The SMD pads [listed at the back of the teensy 4.1](https://www.pjrc.com/teensy/card11b_rev4_web.pdf) isn't included in the library (nor the footprint). Is it possible to include it here?

blackketter, 2021-11-25:
Those pads are designed for SMT parts to be placed on the teensy.

What kind of connector and footprint would you expect on a baseboard to connect to those pins?

mrwm, 2021-11-25:
I was looking to breakout extra SMD pins in the back for I/O as such:

<details><summary>expand to show image</summary>

[image]

</details>

The footprint should be something similar to the ESP-PSRAM32 (which I have pictured above).

mrwm, 2021-11-25:
An example of such usage can be [found with this teensy project](https://www.instructables.com/How-to-Make-a-USB-Laptop-Keyboard-Controller/#step5) that aims to decode laptop keyboard pinouts.

blackketter, 2021-11-25:
Thanks.  I understand how having connections to those pins might be useful, but I'm not sure what connectors could be used on each side to make a reliable connection. Do you have some in mind?

mrwm, 2021-11-25:
As for the footprint, I would assume that it would be ideal to have through hole connectors. With size constrains, leaving them as SMD PSRAM pads would work as well.