XenGi/teensy_library #40: How to convert program code to kicad files

ttttt706, 2022-09-23:
Hello from Japan.
I want to add the .lib and .kicad_mod files of teensy3.5, 
but I don't know how to convert your C programing language(？) to those files.
I’m so sorry trouble you, but please teach me how to do.

XenGi, 2022-09-23:
Hello from Germany!
This repository holds libraries for KiCAD the PCB creation tool. What C files do you mean?

ttttt706, 2022-09-23:
(module Teensy35_36_All_Pins (layer F.Cu) (tedit 6026F021)
  (fp_text reference REF** (at 0 -10.16) (layer F.SilkS)
    (effects (font (size 1 1) (thickness 0.15)))
  )
  (fp_text value Teensy3.5/3.6 (at 0 10.16) (layer F.Fab)
    (effects (font (size 1 1) (thickness 0.15)))
  )
  (fp_poly (pts (xy 1.27 -3.81) (xy 1.524 -3.556) (xy 1.143 -3.302) (xy 0.889 -3.556)) (layer F.SilkS) (width 0.1))
  (fp_poly (pts (xy 1.397 -4.572) (xy 1.651 -4.318) (xy 1.27 -4.064) (xy 1.016 -4.318)) (layer F.SilkS) (width 0.1))
  (fp_poly (pts (xy 1.651 -3.429) (xy 1.905 -3.175) (xy 1.524 -2.921) (xy 1.27 -3.175)) (layer F.SilkS) (width 0.1))
  (fp_poly (pts (xy 1.143 -3.048) (xy 1.397 -2.794) (xy 1.016 -2.54) (xy 0.762 -2.794)) (layer F.SilkS) (width 0.1))
  (fp_poly (pts (xy 1.778 -4.191) (xy 2.032 -3.937) (xy 1.651 -3.683) (xy 1.397 -3.937)) (layer F.SilkS) (width 0.1))
  (fp_poly (pts (xy 0.635 -2.667) (xy 0.889 -2.413) (xy 0.508 -2.159) (xy 0.254 -2.413)) (layer F.SilkS) (width 0.1))
  (fp_poly (pts (xy 0.762 -3.429) (xy 1.016 -3.175) (xy 0.635 -2.921) (xy 0.381 -3.175)) (layer F.SilkS) (width 0.1))
  (fp_poly (pts (xy 1.016 -4.953) (xy 1.27 -4.699) (xy 0.889 -4.445) (xy 0.635 -4.699)) (layer F.SilkS) (width 0.1))
  (fp_line (start -30.48 8.89) (end -30.48 -8.89) (layer F.SilkS) (width 0.15))
  (fp_line (start 30.48 -8.89) (end 30.48 8.89) (layer F.SilkS) (width 0.15))

↑Like this. this is just a example. I found this in the repository for pretty

ttttt706, 2022-09-23:
I am not very familiar with kicad yet. Sorry.

ttttt706, 2022-09-23:
How can I get these codes to load in PCB editor or eeschema?

XenGi, 2022-09-23:
This is how you add footprints:
https://youtu.be/CBfGG5TKc6o

It is the same for libraries. Hope this helps.

ttttt706, 2022-09-23:
Thanks so much! I'll check out this video.

XenGi, 2022-09-23:
Glad I could help.

closed 2022-09-23 by XenGi